pfalcon/picoweb #1: Consider making utemplate imported on demand

dpgeorge, 2015-06-03:
For very simple, light-weight web apps (eg just parse and/or return some json) there is no need for template machinery.  There is (almost) one point where templating is used in picoweb, namely the render_template function.  If it's never called then utemplate need never be imported (and hence take up lots of RAM).  So one can just import utemplate in that funcion and save the template loader object for next time.

pfalcon, 2015-06-04:
I'm ok with this, but any realistic user-facing app would use templates. What's the usercase here? Perhaps, JSON API backend?

dpgeorge, 2015-06-04:
> What's the usercase here? Perhaps, JSON API backend?

Yes, exactly, as mentioned in first post above.

Other option would be "dependency injection" and to provide dummy utemplate.py module in app directory so that was loaded instead.

Reason I didn't have a PR is because I was unsure about "render_str" function and whether you wanted to keep that.

dpgeorge, 2015-06-04:
Let me just say that I originally was going to use http.server module and CGIHTTPRequestHandler class since that's the basic functionality I needed.  But it seemed much easier to just use picoweb, which already had all features plus coroutines (instead of threads in http.server).

Maybe we could promote picoweb as the standard way of doing micro TCP servers in uPy?

Could also make a cutdown version / variant that didn't rely on uasyncio.

pfalcon, 2015-06-04:
> Maybe we could promote picoweb as the standard way of doing micro TCP servers in uPy?

That's certainly how I see it. I didn't look into http.server, but http.client will be a pain to port (need to allow to implement stream protocol in user classes - bunch of bloat for likely the only usecase for the foreseeable future). And yep, with picoweb existing, I'm not really interested to look into http.server. (For the very minimal cases, we have example/unix/http-server.py)

pfalcon, 2015-06-04:
> Other option would be "dependency injection" and to provide dummy utemplate.py module in app directory so that was loaded instead.

Well, one thing a web framework should have (even pico one) is ease of use, and the above is too magic.

But speaking of that, do you also suggest for picoweb to stop depending on utemplate, and let apps which use it pull it instead?

dpgeorge, 2015-06-04:
> But speaking of that, do you also suggest for picoweb to stop depending on utemplate, and let apps which use it pull it instead?

If that's possible then it makes sense.

FYI my json server takes about 39k heap, without utemplate.

pfalcon, 2015-06-06:
>> Maybe we could promote picoweb as the standard way of doing micro TCP servers in uPy?

> That's certainly how I see it. 

Additional point: there's https://github.com/pfalcon/micropython-uaiohttpclient (please don't miss it ;-) ). It's roughly based on original aiohttp API, but again, includes only client, whereas original package has also server support. I don't plan to work on the latter, because again there's picoweb already.

pfalcon, 2015-06-15:
render_str() is there because it's used by notes_pico, i.e. it's straightforward translation of flask API call. Ideally, it should be replaced by generator support in utemplates (i.e. any substitution var should be able to be a generator which renders string content incrementally).

pfalcon, 2015-06-15:
Please see if https://github.com/pfalcon/picoweb/pull/2 does what you need.

dpgeorge, 2015-06-15:
> Please see if #2 does what you need.

It does, thanks!

pfalcon, 2015-06-15:
Merged/published.

closed 2015-06-15 by pfalcon