pfalcon/picoweb #4: cannot upip.install on nodemcu

a-musing-moose, 2016-10-23:
Hi, I'm not sure if this is the correct place to raise this issue so feel free to let me know if now. Thanks for Picoweb, it looks to be exactly what I am after!

I have an ESP8266 flashed with the latest version of Micropython (1.8.5). On a fresh flash attempting to upip install picoweb generates an error:

`
>>> import upip
>>> upip.install('picoweb')
Installing to: /lib/
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "upip.py", line 202, in install
  File "upip.py", line 192, in install
  File "upip.py", line 149, in install_pkg
  File "upip.py", line 138, in get_pkg_metadata
  File "upip.py", line 105, in url_open
OSError: -2
`

If you have any suggestions, or if I am doing something wrong please let me know.

Thanks,
Jon

pfalcon, 2017-01-20:
Sorry, picoweb is (at least so far) not supported on ESP8266 port. However, the error above isn't even related to it, it's a sign of no Internet connectivity. Please direct further discussion of upip on ESP8266 to http://forum.micropython.org/viewtopic.php?f=16&t=2544. Thanks.

closed 2017-01-20 by pfalcon

a-musing-moose, 2017-01-22:
Understood and thanks for the pointer the problem I was actually seeing rather than the one I thought I had.