pfalcon/picoweb #7: How to use with existing uasyncio loop?

craftyguy, 2017-03-07:
I would like to incorporate picoweb into an existing project I have that uses an uasyncio loop for managing several blocking tasks (e.g. buttons), but I see that the WebApp.run implements a new uasyncio loop. Contrary to popular belief, running two of these loops concurrently doesn't seem to be working (I'm a newbie, please don't kill me).

Is the best way around this to have an optional argument to WebApp.run to specify an existing uasyncio loop to attach to and then return, and if an existing loop is not specified then create your own?

Here's a quick and dirty example of what I am talking about, I'm quite sure there's a more elegant way to implement this but I hope it expresses what I think the issue is and how it might be resolved. 

```
    def run(self, host="127.0.0.1", port=8081, debug=False, lazy_init=False, loop=None):
        self.debug = int(debug)
        self.init()
        if not lazy_init:
            for app in self.mounts:
                app.init()
        if debug:
            print("* Running on http://%s:%s/" % (host, port))
        if not loop:
            loop = asyncio.get_event_loop()
            loop.run_forever()
            loop.close()
        else:
            loop.create_task(asyncio.start_server(self._handle, host, port))

```

craftyguy, 2017-03-07:
Ok so I didn't quite understand how uasyncio's "get_event_loop" worked, apparently it's a global handle so my previous modifications are not really needed.

I have tried two different configurations:

1) I extend WebApp.run to just do `loop.create_task` and leave out the run_forever call (since I do that later in my main() function after setting more stuff up)

2) I leave WebApp as is, and just call the `WebApp.run` at the very end of my main() function, after setting up other tasks with `loop.create_task` previously in main()

In either configuration, the following happens:

- picoweb starts the server and when I hit the server with a request from a http client, it never responds to the client. This is a problem.

- other tasks I scheduled and 'sleep' with `mwait asyncio.sleep(N)`, where N is a few hundred seconds, start firing repeatedly after the http request hits picoweb. This is another problem.

Any ideas on how to implement this correctly with picoweb?

craftyguy, 2017-03-07:
Sorry for the comment spam, but I'm managed to reduce this to a really simple test that shows how the other task goes nuts when picoweb receives an http request:

test.py:
```
ROUTES = [
           ("/", lambda req, resp: (yield from picoweb.sendfile(resp, "html/index.html"))),
]

def simple_loop():
    while True:
        print("Yeehaw!")
        yield int(10 * 1000)

def light_alarm():
    loop = asyncio.get_event_loop()
    loop.create_task(simple_loop())
    webapp = picoweb.WebApp(__name__, ROUTES)

```

This outputs:
```
>>> light_alarm.light_alarm()
* Running on http://0.0.0.0:80/
Yeehaw!
pm open,type:2 0
Yeehaw!       <------------ http client connected at this point, remaining messages came out in about a 2s interval
Yeehaw!
Yeehaw!
542213010.787 <HTTPRequest object at 3fff3cd0> <StreamWriter <socket state=2 timeout=0 incoming=3fff90a0 off=26>> "GET /index.html"
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
542213010.787 <HTTPRequest object at 3fff4f60> <StreamWriter <socket state=2 timeout=0 incoming=3fff90a0 off=26>> "GET /styles.css"
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
542213010.787 <HTTPRequest object at 3fff57d0> <StreamWriter <socket state=2 timeout=0 incoming=3fff90a0 off=27>> "GET /favicon.ico"
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!
```

I added a note in the output where the http client connected to picoweb and the other task seemed to be completely ignoring the yield and printed to console a bunch of times.

pfalcon, 2017-03-07:
@craftyguy : You're welcome to pose good questions and then try to find answers for them in a way which may be useful to other people (using "comment spam" and stuff, I like that). I don't have a good answer to your question(s) right away, but can says there're few known issues with uasyncio scheduling which gets tackled literally as we speak, like couple of things were fixed just last couple of days, and there're more to fix/implement, see e.g. https://github.com/micropython/micropython/pull/2909. It may take some time for me to look in detail at what you wrote, and even more to make uasyncio to do what I have in mind, and then even more to look into others' people features, but there's sustainable movement that way, and again, you're welcome to provide good feedback (like, based on your trying to do something).

craftyguy, 2017-03-07:
@pfalcon Thank you for the encouragement :) I'm almost positive my issue stems from my very limited understanding of micropython and uasyncio, combined with my relatively novice experience with scheduling on Python. I'll continue to poke through the source code for uasyncio, picoweb, and keep all components up to date with the latest in git!

bmsleight, 2017-06-06:
@craftyguy I am in the same place as you (very limited understanding of micropython and uasyncio, combined with my relatively novice experience with scheduling on Python).

With a daily build of firmware [MicroPython v1.9-25-g9db1c50f on 2017-06-06; ESP module with ESP8266]
Starting with https://github.com/pfalcon/picoweb/blob/master/example_webapp.py and using some of the code above....

```
#
# This is a picoweb example showing a centralized web page route
# specification (classical Django style).
#
import ure as re
import picoweb
import uasyncio as asyncio

def simple_loop():
    while True:
        print("Yeehaw!")
        yield int(10 * 1000)

def index(req, resp):
    # You can construct an HTTP response completely yourself, having
    # a full control of headers sent...
    yield from resp.awrite("HTTP/1.0 200 OK\r\n")
    yield from resp.awrite("Content-Type: text/html\r\n")
    yield from resp.awrite("\r\n")
    yield from resp.awrite("I can show you a table of <a href='squares'>squares</a>.<br/>")
    yield from resp.awrite("Or my <a href='file'>source</a>.")

def squares(req, resp):
    # Or can use a convenience function start_response() (see its source for
    # extra params it takes).
    yield from picoweb.start_response(resp)
    yield from app.render_template(resp, "squares.tpl", (req,))

def hello(req, resp):
    yield from picoweb.start_response(resp)
    # Here's how you extract matched groups from a regex URI match
    yield from resp.awrite("Hello " + req.url_match.group(1))

ROUTES = [
    # You can specify exact URI string matches...
    ("/", index),
    ("/squares", squares),
    ("/file", lambda req, resp: (yield from app.sendfile(resp, "example_webapp.py"))),
    # ... or match using a regex, a match result available as req.url_match
    # for group extraction in your view.
    (re.compile("^/iam/(.+)"), hello),
]

import logging
logging.basicConfig(level=logging.INFO)
#logging.basicConfig(level=logging.DEBUG)

loop = asyncio.get_event_loop()
loop.create_task(simple_loop())
app = picoweb.WebApp(__name__, ROUTES)
app.run(host="0.0.0.0", debug=True)

```
Now gives:-

```
>>> import example_webapp
* Running on http://0.0.0.0:8081/
Yeehaw!
25.000 <HTTPRequest object at 3fff3d40> <StreamWriter <socket state=2 timeout=0 incoming=3fff9aa0 off=16>> "GET /"
Yeehaw!
29.000 <HTTPRequest object at 3fff5260> <StreamWriter <socket state=2 timeout=0 incoming=3fff9aa0 off=16>> "GET /"
33.000 <HTTPRequest object at 3fff2f40> <StreamWriter <socket state=2 timeout=0 incoming=3fff9aa0 off=16>> "GET /"
35.000 <HTTPRequest object at 3fff6420> <StreamWriter <socket state=2 timeout=0 incoming=3fff9aa0 off=16>> "GET /"
Yeehaw!
Yeehaw!
Yeehaw!
Yeehaw!

```

No more going nuts ? 
Maybe close this bug?

It gave me a very good start. Thanks.
BMS

craftyguy, 2017-06-06:
Great! Yea I've since learned quite a bit about how to use uasyncio, though I no longer need picoweb. I may need it again in the future :)

closed 2017-06-06 by craftyguy