pfalcon/picoweb #11: stream writer bigger than buffer

murilobsd, 2017-06-06:
Hi, congratulations on the great work with picoweb. 

I'm trying to use the r**ender_template** method and an _AssertionError_ error is occurring, in the _uasyncio_ library more precisely in the **awrite** method, there is a check **of the size of the buffer size of the StreamWriter** which occurs is that the template I'm trying to render is larger than the buffer , Is there any way around this?

Size Stream: 2456 bytes
Buffer: 2455 bytes

> * Running on http://127.0.0.1:8081/                                                   
Size Stream: 2457 and size buffer: 2455 
Traceback (most recent call last):                                                     
  File "server.py", line 16, in <module>                                             
  File "/home/murilobsd/.micropython/lib/picoweb/__init__.py", line 240, in run      
  File "/home/murilobsd/.micropython/lib/uasyncio/core.py", line 121, in run_forever  
  File "/home/murilobsd/.micropython/lib/uasyncio/core.py", line 85, in run_forever    
  File "/home/murilobsd/.micropython/lib/picoweb/__init__.py", line 156, in _handle   
  File "server.py", line 13, in home                                                  
  File "/home/murilobsd/.micropython/lib/picoweb/__init__.py", line 196, in render_template                                                                               
  File "/home/murilobsd/.micropython/lib/uasyncio/__init__.py", line 155, in awrite   
AssertionError:

I'm using:

- ESP8266
- MicroPython v1.9-21-ga0dbbbeb-dirty on 2017-06-04; ESP module with ESP8266
- picoweb: 1.1.1
- uasyncio: 1.1.2
- utemplate: 1.1

pfalcon, 2017-06-06:
If you look at that line (btw, somehow it's line 156 in the current micropython-lib), it's implausible to fail. So, please print the actual values used in the comparison in that assert.

And to investigate/fix it, I'd need a minimal testcase and steps to reproduce the issue. (Minimal here would mean using only uasyncio, as the error happens in it, not in picoweb.)

pfalcon, 2017-06-07:
I couldn't reproduce this issue on uasyncio level. I also took a chance to optimize how .awrite() works, please retest with new version: https://github.com/micropython/micropython-lib/commit/75e1474ddfed408f4b01c8abf33aa059967ecf7c

murilobsd, 2017-06-07:
@pfalcon I'm sorry for not having passed the values, basically the test:

line: 155 uasyncio
assert res < sz 

res = 919 
sz = 917

I will perform a new test, thank you.

murilobsd, 2017-06-07:
@pfalcon after updated, when I request the GET method it works! But when I request the POST method not. See an example:

```python
@app.route("/test")
def test(req, resp):
    yield from picoweb.start_response(resp)
    req.read_form_data()
    yield from app.render_template(resp, "test.tpl")

```

```bash
$ curl -d "param1=value1&param2=value2" -X POST http://localhost:8081/test

.......

curl: (56) Recv failure: 'Connection reset by peer'
```

````
================               
<HTTPRequest object at 14e6fbe0> <StreamWriter <_socket 4>>                                                                                    
<HTTPRequest object at 14e6fbe0> ('POST', '/teste', '', 'HTTP/1.1') {'Accept': '*/*', 
'User-Agent': 'curl/7.54.0', 'Host': 'localhost:8081', 'Content-Length': '27', 'Content-Type': 'application/x-www-form-urlencoded'}                     
                                                                 
{'headers': {'Accept': '*/*', 'User-Agent': 'curl/7.54.0', 'Host': 'localhost:8081', 'Content-Length': '27', 'Content-Type': 'application/x-www-form-urlencoded'}, 'qs': '','path': '/teste', 'method': 'POST', 'reader': <StreamReader <_socket 4>>}      
```

pfalcon, 2017-06-07:
Your response doesn't provide enough information - you don't even show what error happens on MicroPython side, and as I mentioned, in general, a minimal complete example to reproduce is needed.