pfalcon/picoweb #12: Error installing 'micropython-uasyncio': , packages may be partially installed

bmsleight, 2017-06-13:
Was working a few days ago...

```
MicroPython v1.9.1-8-g7213e78d on 2017-06-12; ESP module with ESP8266
Type "help()" for more information.
>>> import upip
>>> upip.install("picoweb")
Installing to: /lib/
Warning: pypi.python.org SSL certificate is not validated
Installing picoweb 1.2 from https://pypi.python.org/packages/07/a1/ac79de53d38d6ae531c857f3d9291afaa27d90c58eca9a1d295d99f7546a/picoweb-1.2.tar.gz
Error installing 'micropython-uasyncio': , packages may be partially installed
>>> import uasyncio
>>> import picoweb
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/picoweb/__init__.py", line 8, in <module>
ImportError: no module named 'pkg_resources'

```

zjor, 2017-06-29:
I'm facing the same issue

pfalcon, 2017-06-29:
Unless you can provide details showing otherwise, the above is a transient networking error, just retry.

zjor, 2017-06-29:
I figured out what was the problem, `pkg_resources` should be installed as
```
micropython -m upip install micropython-pkg_resources
```

pfalcon, 2017-06-29:
I see, this issue is fixed in https://github.com/pfalcon/picoweb/commit/3508517a824b891e29dedacabef2e9e2cc144f57

closed 2017-06-29 by pfalcon