pfalcon/picoweb #13: failed to install on micropython-esp using upip

fruch, 2017-07-24:
I'm really new to micropython and trying to install this package with upip:

    import upip
    upip.install('picoweb')

and it keep failing saying the package is partly installed

pfalcon, 2017-10-31:
Try after reset.

larsks, 2017-11-10:
@pfalcon I thought that picoweb didn't work on esp8266 boards (#4, #9)? Or has that changed? Because if it has, I'm having the same problem as @fruch.

pfalcon, 2017-11-10:
@larsks: See https://github.com/pfalcon/notes-pico#deploying-on-embedded-micropython-targets

larsks, 2017-11-10:
@pfalcon Thanks for the reference!

flyboy74, 2017-11-20:
I am trying to get picoweb running on my ESP8266. I was quite sure how to follow pfalcon link.
I installed esp8266-20171101-v1.9.3.bin and have connected via serial tt to micropython and connected to my Wifi router and started WebREPL successfully. But have been unable to install Picweb. Do I need a different firmware??

fruch, 2017-11-20:
@flyboy74 follow the instructions from @pfalcon notes. 

thing is that the esp8266 doesn't have enough memory for the upip to download "big" packages correctly. (big is not the files, but responses from pypi)

the best thing todo is to download micropython source and build the need packages into the image.

flyboy74, 2017-11-20:
Thanks I now understand the problem. Now for me to understand the work around. I used my windows PC to flash the BIN file to the ESP8266 using ESP8226flasher but I do also have esptool.py downloaded. All the instructions seem to be for a linux machine. Can I install picoweb from my windows machine??

flyboy74, 2017-11-20:
I do own Raspberry pi. Will it be easier to install from my RPi

flyboy74, 2017-11-20:
Also I was wondering if picoweb can be included in standard installation of micropython for ESP chips considering the purpose that ESP chip are used for

fruch, 2017-11-20:
I would recommend building on a VM (install virtualbox) or on a docker instance. maybe things in win10 improved and you can use the new linux distro in windows, but I never used it, so can't say how easy it is to work with it.

flyboy74, 2017-11-20:
sounds hard especially since my SSD drive is nearly full on my PC. I have ordered a ESP32 dev board and I believe it has the capacity to install picoweb using upip. I might have to wait till it arrives might be easier.

flyboy74, 2017-11-20:
Where's the best place for me to raise the issue of having picoweb prebuilt into the micropyton BIN file for the ESP boards especially the ESP8266

pfalcon, 2018-01-30:
> Where's the best place for me to raise the issue of having picoweb prebuilt into the micropyton BIN file for the ESP boards especially the ESP8266

There're (hopefully) thousands of MicroPython users and each of them wants their favorite modules to be embedded. However, that doesn't scale: if we add all those modules, the binary won't fit on any board out there. That's why (Micro)Python uses modular approach instead. Yes, using it requires some learning, but that's true about anything in Computer Science/Information Technologies. MicroPython is open-source project, so it's important that people actually exercised their right to use and deal with source, or it's all failure. Please don't make our beloved project a failure ;-).

Btw, there's now summarized docs on dealing with various aspects of uPy application deployment: http://pycopy.readthedocs.io/en/latest/reference/packages.html .

pfalcon, 2018-01-30:
Other, the original issue in the title doesn't seem to be reproducible (but certainly can be a transient failure).

So, I'm closing this, feel free to open a new ticket with detailed information about your setup and steps to reproduce.

closed 2018-01-30 by pfalcon