pfalcon/picoweb #14: how to handle OSError: [Errno 104] ECONNRESET

fruch, 2017-07-24:
how can I handle those, gracefully ?

Traceback (most recent call last):
  File "main.py", line 69, in <module>
  File "/lib/picoweb/__init__.py", line 240, in run
  File "uasyncio/core.py", line 121, in run_forever
  File "uasyncio/core.py", line 85, in run_forever
  File "/lib/picoweb/__init__.py", line 156, in _handle
  File "main.py", line 47, in <lambda>
  File "/lib/picoweb/__init__.py", line 214, in sendfile
  File "/lib/picoweb/__init__.py", line 214, in sendfile
  File "/lib/picoweb/__init__.py", line 209, in sendfile
  File "/lib/picoweb/__init__.py", line 209, in sendfile
  File "/lib/picoweb/__init__.py", line 28, in sendstream
  File "uasyncio/__init__.py", line 144, in awrite
OSError: [Errno 104] ECONNRESET

pfalcon, 2017-11-01:
Needs wrapping in try/except, I need to look into that...

As a workaround, you can try to do that on your side.

pfalcon, 2017-12-01:
With https://github.com/pfalcon/picoweb/commit/dd579fd6ba609e6c313b8c967a4b2fd7e5942972 , Picoweb no longer dies if there's an error in a single request served, but continues.

closed 2017-12-01 by pfalcon