pfalcon/picoweb #16: ESP32 support

anteph, 2017-09-01:
Hi! 

I don't know if this is the best place to share, but I've been able to install and use the Picoweb on an ESP32. 

I'm leaving the instructions that worked for me bellow, in case someone else wants to give it a try.
https://techtutorialsx.com/2017/09/01/esp32-micropython-http-webserver-with-picoweb/

Nonetheless, I've not been able to use the temple features, due to unavailability of memory to allocate. But running the server and creating some routes works pretty well.

Thanks for your great work on this module.

Best regards,
Nuno Santos

pfalcon, 2017-10-31:
@anteph : Thanks for the note! Regarding template issues, can leave more comments so other people could compare their results? Did you try https://github.com/pfalcon/picoweb/blob/master/example_webapp.py for example? ESP8266 is known to run an even larger application with templates, so they should work ESP32 (unless it still has some issues; I don't test on ESP32 myself so far).

anteph, 2017-11-05:
Hi! 

At the time I've tested I didn't notice that we could install picoweb via upip, so I installed it manually, which may have caused the issue. I was experiencing the issues with the templates basically by running the example application you mention.

I haven't yet had time to test again the templates with a clean installation via upip, but as soon as I have the opportunity I will do it and share the results here.

Best regards,
Nuno Santos