pfalcon/picoweb #17: Picoweb and temperature read in the loop (uasyncio)

mvincm, 2017-09-14:
Hello,

As there is no forum or blog comments I must use this kind of public commutation chanel ;)

First of all - great job ! Thanks !

I would like to use ESP32 (with micropython) and picoweb to made some thermostat. I will get and set temperature via picoweb "web app". To communicate I will use simple json messages... but I need to check temperature every e.g. 2 minutes, compare it stored one and turn on or off my heater. 

So my idea is:
- in file "__init__.py" I will add second scheduled  job:

```
loop.create_task(asyncio.start_server(self._handle, host, port))
loop.create_task(check_temperature())
```

- then define this function (outside WebApp class) and put some code to read temperature and turn on/off heater

```
async def check_tempersture():
      while True:
        #check temperature
        # some IFs
        # turn on/off heater
        await asyncio.sleep(120)
```

And the question is... Is this the right way to do this? Any better solution?

Best regards,
MvincM

dpgeorge, 2017-09-15:
> As there is no forum or blog comments I must use this kind of public commutation chanel ;)

If you like you can discuss these things at the MicroPython forum: https://forum.micropython.org, in particular there is an esp32 sub-forum at https://forum.micropython.org/viewforum.php?f=18

mvincm, 2017-09-17:
Thanks. I will look at this forum.

M

pfalcon, 2017-10-31:
> in file "init.py" I will add second scheduled job:

A natural way to do it is to subclass WebApp class and override init() method to add things you need.

pfalcon, 2017-12-01:
An example was added to the repo showing how to do that (and more): https://github.com/pfalcon/picoweb/blob/master/examples/example_eventsource_push.py

There's very little magic: you just get an event loop and .create_task() in it. Then it runs when Picoweb app is started.

closed 2017-12-01 by pfalcon