pfalcon/picoweb #25: Support encoded content

orzechow, 2017-12-31:
I'd like to send static content with gzip encoding (see [MDN dev docs](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Content-Encoding#Compressing_with_gzip)) in order to increase the page load speed.

So ideally `sendfile()` 
https://github.com/pfalcon/picoweb/blob/6e3685024968624337e20bd48f42e48abf4371c0/picoweb/__init__.py#L238
provides an argument for the content encoding:
```python
def sendfile(self, writer, fname, content_type=None, content_encoding=None):
```
as well as `start_response()`:
https://github.com/pfalcon/picoweb/blob/6e3685024968624337e20bd48f42e48abf4371c0/picoweb/__init__.py#L39
```python
def start_response(writer, content_type="text/html", status="200", content_encoding=None):
```

I don't want to gzip the files on the fly, but provide already gzipped content instead of raw content if available. An example for css files:
```python
@app.route(re.compile('^\/(.+\.css)$'), methods=['GET'])
def styles(request, response):
    file_path = request.url_match.group(1)

    if b"gzip" in request.headers[b"Accept-Encoding"]:
        file_path_gzip = file_path + ".gz"
        import os
        if file_path_gzip in os.listdir("/"):
            yield from app.sendfile(response, file_path_gzip, 'text/css', 'gzip')
            return

    yield from app.sendfile(response, file_path, 'text/css')
```

orzechow, 2017-12-31:
As this should not be a big thing and I'm working towards it, I'll implement this in a fork and send a merge request.

orzechow, 2017-12-31:
Well I just found pull request #5, so this is a duplicate.

Though I would propose to leave the part of providing the encoded content to the developer as shown in my example. This would keep picoweb itself minimal, but still provide encoding to those who want to use it.

pfalcon, 2017-12-31:
You're welcome to. But please don't repeat mistakes in https://github.com/pfalcon/picoweb/pull/5 : Picoweb is **the** minimalist framework, so any byte of the code added (and any memory allocation done by it) goes against its nature. (Which is not the reason to not do useful things, it's the reason to do it right and minimal.)

pfalcon, 2018-01-30:
Was implemented in https://github.com/pfalcon/picoweb/pull/27 , thanks!

closed 2018-01-30 by pfalcon