pfalcon/picoweb #26: Add Cache-Control header

orzechow, 2017-12-31:
The HTML header "Cache-Control" might be useful for bigger static content.

A minimal version has already been implemented in https://github.com/joewez/picoweb/pull/1

I propose an even more minimal approach that does not include his changes in `get_mime_type()`:
```python
def sendfile(self, writer, fname, content_type=None, max-caching-age=0):
```

Merge request will follow.

pfalcon, 2017-12-31:
So, from #25 and this ticket it should be clear that there's one (or two) little functions and too many headers people want to stuff into them (more were asked at https://forum.micropython.org/viewtopic.php?f=15&t=165). Consequently, they should just take a dictionary with "extra headers".

pfalcon, 2017-12-31:
Actually, should take a dictionary or just bytes of preformatted headers, for efficiency.

orzechow, 2017-12-31:
True, I guess an "extra headers" argument would be sufficient and a good trade-off.

pfalcon, 2018-01-30:
Arbitrary headers now can be sent with a response.

closed 2018-01-30 by pfalcon