pfalcon/picoweb #31: ESP32 giving failure with picoweb

darksoal, 2018-01-24:
Hey guys!

I'm trying to setup my ESP32 to run picoweb. I did a fresh installations of Micropython, did all the installs with upip. but when I try to run the example code i get the following response: 

Traceback (most recent call last):
  File "main.py", line 64, in <module>
  File "/lib/picoweb/__init__.py", line 287, in run
  File "/lib/uasyncio/core.py", line 146, in run_forever
  File "/lib/uasyncio/core.py", line 101, in run_forever
  File "/lib/uasyncio/__init__.py", line 235, in start_server
TypeError: function takes 2 positional arguments but 4 were given

the code I'm currently running is: 

```
import uasyncio
import picoweb

def index(req, resp):
    yield from picoweb.start_response(resp)
    yield from resp.awrite("""\
<!DOCTYPE html>
<html>
<head>
<script>
var source = new EventSource("events");
source.onmessage = function(event) {
    document.getElementById("result").innerHTML += event.data + "<br>";
}
source.onerror = function(error) {
    console.log(error);
    document.getElementById("result").innerHTML += "EventSource error:" + error + "<br>";
}
</script>
</head>
<body>
<div id="result"></div>
</body>
</html>
""")

def events(req, resp):
    print("Event source connected")
    yield from resp.awrite("HTTP/1.0 200 OK\r\n")
    yield from resp.awrite("Content-Type: text/event-stream\r\n")
    yield from resp.awrite("\r\n")
    i = 0
    try:
        while True:
            yield from resp.awrite("data: %d\n\n" % i)
            yield from uasyncio.sleep(1)
            i += 1
    except OSError:
        print("Event source connection closed")
        yield from resp.aclose()

ROUTES = [
    ("/", index),
    ("/events", events),
]

app = picoweb.WebApp(__name__, ROUTES)
app.run(debug=True)
```
my network configuration is done in a boot file before this code is being run but i don't think the problem is in there. Any of you have an idea where I might have made an error?

pfalcon, 2018-01-25:
You should follow the stack you got and check the code located there. You will that the problem happens with getaddrinfo() function, which gets passed 4 args, but esp32 port accepts only 2. This problem should be reported to esp32 port for fixing.

pfalcon, 2018-01-27:
I've applied workaround in my repository: https://github.com/pfalcon/micropython/commit/d4803e831492cdc206a4fc3c3145f2b3ebdfe2b4 . The repo will be rebased, link will stop to work, just use https://github.com/pfalcon/micropython . Note that uasyncio, on which picoweb is built, in general requires my fork of MicroPython.

pfalcon, 2018-02-24:
This should be fixed in micropython mainline now.

closed 2018-02-24 by pfalcon