pfalcon/picoweb #39: Install picoweb in nodemcu

PabloF7, 2018-06-24:
Hi, is it possible to install picoweb in nodemcu board? I run
`upip.install('picoweb')`

but i got:
`Installing to: /lib/  Warning: pypi.org SSL certificate is not validated
Error installing 'picoweb': memory allocation failed, allocating 7406 bytes, packages may be partially installed`

Any tips? thanks in advance.

lingster, 2018-08-24:
Hi, you will have to create your own micropython firmware with picoweb frozen/included in the bin file, (there's not enough memory to be able to download from pypi to your device) . Then you will need to reflash your device. Here's and article on how to do so:

https://learn.adafruit.com/micropython-basics-loading-modules/frozen-modules

I have just been down this path with my esp01, you will need to build a few tools and run a few make files along the way, but it does work a treat!

pfalcon, 2018-08-24:
> Hi, is it possible to install picoweb in nodemcu board?

It's possible, but it's a low-memory board, so you can easily run out of memory. It's recommended to run "upip" immediately after reset.

Using frozen modules is indeed another approach.

closed 2018-08-24 by pfalcon

allyourcode, 2020-05-07:
So.. I've figured out frozen modules in general (Thank you for the pointer, @lingster !), but not sure about how to do that for picoweb specifically... What I've tried to do is simply copy the "picoweb" dir from this repo into the micropython/.../modules dir and building micropython with the extra files (from picoweb). The build succeeds, and I am able to load it onto my board, but when I try to import picoweb, I get a(n unhandled) exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "picoweb/__init__.py", line 12, in <module>
ImportError: no module named 'pkg_resources'

Not sure what that means :/ It looks like it's related to libraries that picoweb needs. Anyway... Help??

allyourcode, 2020-05-07:
Ok, I hacked my way around the problem by commenting out line 12. It looks like pkg_resources is only used in one place anyway. Seems like we ought to rip out that dependency, since pkg_resources isn't available in Micropython. I didn't look too carefully at how picoweb uses pkg_resources, but it seems like the functionality could be "in-lined" into picoweb itself (i.e. picoweb could include its own implementation).

Once I got past that, I ran into another issue: picoweb imports ulogging, but there is no such module in Micropython. (also, the import occurs within a function, not at the top of the __init__.py file, where it would traditionally be found). I got around that by passing debug=-1 to app.run, but it seems to me that a) the import should be moved to the top b) if logging is not available (such as in Micropython), and debug=0 or greater is passed to app.run, then an exception should be raised. Perhaps, it would be of type NotImplementedError?

allyourcode, 2020-05-07:
Oh, and another trap for young players: pass host='0.0.0.0', port=80 to app.run. Otherwise, you will be sad and confused when your browser cannot talk to your web server!