pfalcon/picoweb #40: cannot read form data on ESP8266

shashikantx, 2018-10-21:
I have been trying to read post data,

My code in MicroPython is 

`import picoweb
import uasyncio as asyncio

def info(req, resp):
   
    yield from picoweb.start_response(resp)
    yield from resp.awrite("Alrught")
    req.read_form_data()
    print(req.form)

ROUTES = [
    ("/info", info),
]

import logging
logging.basicConfig(level=logging.INFO)

app = picoweb.WebApp(__name__, ROUTES)

app.run(port=80,host="0.0.0.0",debug=1)
`

the server runs without hassles,
but problem is I am unable read post data, the error I keep getting is,

`AttributeError: 'HTTPRequest' object has no attribute 'form' `

I tried the **parse_qs()** and then access **form** it works, but same won't work for **read_form_data()**

I don't understand what is going wrong.

pfalcon, 2018-11-15:
`read_form_data()` is a coroutine, so you should call it like `yield from req.read_form_data()` or `await req.read_form_data()`.

There was also no example on how to work with forms, now I added one: https://github.com/pfalcon/picoweb/blob/master/examples/example_form.py

pfalcon, 2018-11-15:
Btw, thanks for taking care to format the code snippet here well, appreciated. But you don't need to put every single line in backticks, instead just wrap the whole code in triple backticks, like:

~~~
```
your
code
and
more
```
~~~

closed 2018-11-15 by pfalcon