pfalcon/picoweb #44: Connection aborted with large body

SimoneS93, 2018-11-18:
I'm using picoweb on a ESP32 to share files between boards and I'm experimenting with the max body size it will handle. Of course I'm reading the request in chunks with the following code:

```
@app.route("/post")
def post(req, res):
  print("post")
  remaining = int(req.headers[b"Content-Length"])
  print("length", remaining)
  chunk = 128
  file = open("www/post.txt", "w")
  while remaining > 0:
    print("remaining...", remaining)
    to_read = min(remaining, chunk)
    data = yield from req.reader.read(to_read)
    file.write(data)
    remaining -= to_read
  file.close()
  del data
  print("OK")
  yield from picoweb.start_response(res)
  yield from res.awrite("OK")
```

With large body (15 to 50 kb), the REPL prints "OK" and file gets written well, but the connection gets aborted and the response from curl is

`curl: (56) Recv failure: Connection aborted from sender` (translated from italian " Connessione interrotta dal corrispondente")`

Of course I would like to get an "OK" response: does anyone have suggestions?

pfalcon, 2018-11-18:
This doesn't provide enough info to tell what exactly happens. By calling for some telepathy, let me guess that you get ECONNRESET error. You should read up on it to understand better how TCP connections should be handled.

Following that hypothesis, the problem would be here:

~~~
    data = yield from req.reader.read(to_read)
    file.write(data)
    remaining -= to_read
~~~

You here assume that .read() returns as much data as you asked for. This is *never* the case, it may return less data at any time.

SimoneS93, 2018-11-19:
You were right, I solved replacing

`remaining -= to_read` with `remaining -= len(data)`.

Now I can send even 120Kb without problem. Thank you.

closed 2018-11-19 by SimoneS93