pfalcon/picoweb #49: How to add image to the page?

andras-gyarmati, 2019-02-07:
Hi,

Please provide an example of hosting a picture on the website. I could not find any reference on google/github/blogpost how to do it. 

Thank you!

andras-gyarmati, 2019-02-07:
I just figured out. Put an img tag in the template:
```
<img src="templates/your_image_name.jpg"/>
```
and then you serve the request:
```
@app.route("/your_image_name.jpg", methods=['GET'])
def your_method_name(req, resp):
    cleanup()
    yield from picoweb.start_response(resp)
    try:
        with open("templates/your_image_name.jpg", 'rb') as img_binary:
            img= img_binary.read()
        yield from resp.awrite(img)
    except Exception:
        print("Image file not found.")
        pass
    cleanup()
```

pfalcon, 2019-02-07:
It's even easier than that. An image file is no different than any other static file. There're example showing how to serve static files (search for "sendfile"), but it's even easier than that: any webapp is by default pre-configured to serve static files from webapps's `static/` folder. Just put your stuff there (images/whatever) and refer to them as `static/...` in your HTML.

I recommend looking at a "real-world" example of app at https://github.com/pfalcon/notes-pico/tree/master/notes_pico .

And yeah, I'll add an image example when a have some time.

pfalcon, 2019-06-17:
Example on how to serve images - both static and dynamically-generated once - is added at https://github.com/pfalcon/picoweb/blob/master/examples/example_img.py .

closed 2019-06-17 by pfalcon