pfalcon/picoweb #50: Does picoweb support async handlers? If so, document this

buzzware, 2019-05-06:
I want to do some async operations in a handler eg. call an API, process and return the result. Would picoweb support something like :

```
@app.route("/temp")
async def temp(req, resp):       # an async handler
    temp = await httpClient.get('/status')
    yield from picoweb.start_response(resp)
    yield from resp.awrite("Temperature: "+temp['temperature'])
```

If so, please add an async example. The word async hardly exists in the whole repository.
Apologies if I'm missing something, I'm new to Micropython.

buzzware, 2019-05-06:
This suggests that async handlers might work, because "In MicroPython coroutines and generators are identical" and "The await bar() syntax is equivalent to yield from bar()"
https://github.com/peterhinch/micropython-async/blob/master/UNDER_THE_HOOD.md#2-generators-and-coroutines

pfalcon, 2019-05-06:
Please see https://www.python.org/dev/peps/pep-0492/ and Python (not even MicroPython!) documentation for answers. Short summary: "await" is a novelty, syntactic sugar for "yield from".

Picoweb is built on top of uasyncio and thus a fully async web framework. Please see existing documentation and examples for more information.

closed 2019-05-06 by pfalcon

buzzware, 2019-05-12:
Its quite complex to understand async/await, generators, async generators, uasyncio/asyncio and micropython vs python.
I don't yet fully understand how await == "yield from" or "In MicroPython coroutines and generators are identical" vs the situation in CPython. Any good references?
I tried making my handler async and it did seem to work. I'll report back if I hit any issues.
I still think an async example or documentation mention would help.