pfalcon/picoweb #51: Server often crashes with 'NameError: local variable referenced before assignment'

osechet, 2019-05-06:
Rather often, my picoweb server crashes with the following traceback:
```
Traceback (most recent call last):
  File "main.py", line 39, in run
  File "/lib/picoweb/__init__.py", line 299, in run
  File "/lib/uasyncio/core.py", line 163, in run_forever
  File "/lib/uasyncio/core.py", line 116, in run_forever
  File "/lib/picoweb/__init__.py", line 210, in _handle
  File "/lib/picoweb/__init__.py", line 210, in _handle
NameError: local variable referenced before assignment
```
I added some logs to understand and the problem happens when calling uayncio's `readline`, at the beginning of the function. The original error is:
```
Traceback (most recent call last):
  File "/lib/picoweb/__init__.py", line 116, in _handle
  File "/lib/uasyncio/__init__.py", line 131, in readline
AssertionError: 
```
When `readline` is called, the `req` variable is not initialized. It explains the 'local variable referenced before assignment' error.

I moved the `req = HTTPRequest()` at the beginning of the try block to avoid the error. I now get the error log and my server does not crashes. There is maybe a better solution.

osechet, 2019-05-06:
I forgot to mention I'm using picoweb 1.5.3. It's running on a pycopy I built yesterday.

pfalcon, 2019-05-06:
> I moved the req = HTTPRequest() at the beginning of the try block to avoid the error. I now get the error log and my server does not crash.

Good catch. I opted for https://github.com/pfalcon/picoweb/commit/c04021571700c624dc2db0e18efb6945e6f3e402 fix though.

>   File "/lib/uasyncio/__init__.py", line 131, in readline
AssertionError: 

I'm not sure why you were hitting that assert, you should not hit it with normal TCP (e.g. HTTP) connections. But it might hit with wensockets and upcoming TLS support, so I'm fixing it too: https://github.com/pfalcon/micropython-lib/commit/f9fef08913c05bc69d3de27bce72805da91b648a .

Feel free to retest everything with these latest versions (both released on PyPI) - as I say, it's a bit strange you were hitting those asserts at all.

Thanks for the report!

closed 2019-05-06 by pfalcon

osechet, 2019-05-07:
@pfalcon I just tested the latest version of usayncio. It's working. Thanks. 

For the reason of the "invalid" request, if it can help, the call was made from Angular's [HttpClient](https://angular.io/guide/http).

pfalcon, 2019-05-07:
> I just tested the latest version of usayncio. It's working.

Thanks for confirming!