pfalcon/picoweb #52: Unicode characters in template problem

tarakanov, 2019-06-12:
When I use unicode characters in template, a browser receive distorted chars. Using chars as bytes in template gives me an error. It also distorted when I send `resp.write('абвгд')` directly, without template. 

I can send correctly using bytes  - `resp.write(b'абвгд')`. So I rewrite `render_template`:
```
def render_template_b(writer, template_name, args=()):
  import utemplate.source
  template_loader = utemplate.source.Loader("__main__", "templates")
  template = template_loader.load(template_name)
  for s in template(*args):
    yield from writer.awrite(s.encode('utf-8'))
```
Is it a bug? Or how can I use unicode chars in template?

pfalcon, 2019-06-12:
> I can send correctly using bytes - resp.write(b'абвгд')

Yes, that's the correct way to do it - .write() accepts `bytes` object. MicroPython is more relaxed than CPython and allows to pass `str` in many places, but that may not work reliably for non-ASCII chars, as you noticed.

The change you made to render_template() looks plausible, but I would like to consider any other possible alternatives. Can you provide small self-contained example which reproduces the problem to speed up the resolution?

tarakanov, 2019-06-12:
Here is the script:
```
import picoweb

app = picoweb.WebApp(__name__)

def render_template_b(writer, template_name, args=()):
  import utemplate.source
  template_loader = utemplate.source.Loader("__main__", "templates")
  template = template_loader.load(template_name)
  for s in template(*args):
    yield from writer.awrite(s.encode('utf-8'))

@app.route("/ascii")
def index(req, resp):
    yield from picoweb.start_response(resp, content_type = "text/html")
    yield from app.render_template(resp, "ascii.tpl", ("some ascii chars",))

@app.route("/non-ascii-fix")
def test(req, resp):
    yield from picoweb.start_response(resp, content_type = "text/html; charset=utf-8")
    yield from render_template_b(resp, "non-ascii.tpl", ("абвгдe",))

@app.route("/non-ascii-error")
def test2(req, resp):
    yield from picoweb.start_response(resp, content_type = "text/html; charset=utf-8")
    yield from app.render_template(resp, "non-ascii.tpl", ("абвгдe",))

import ulogging as logging
logging.basicConfig(level=logging.INFO)

```

`ascii.tpl`:
```
{% args chars %}
<html>
ascii chars: abcde <br>
From args: {{ chars }} <br>
</html>
````
`non-ascii.tpl`:
```
{% args chars %}
<html>
Non ascii chars: абвгде<br>
From args: {{ chars }} <br>
</html>
```

pfalcon, 2019-06-13:
Ok, so my thoughts about this stuff:

1. Ideally, I'd like .awrite() to just work with `str`.
2. However, I'm a bit shy to add `if isinstance(buf, str): buf = buf.encode()` to it, conceptually it's "wrong" to make "well-behaving" apps, which shove the `bytes` in, to incur overhead of that check.
3. Then patching it on the level of render_template(), like proposed here would be the way, right?
4. No, wait there's p.1.
5. Looking at StreamWriter.awrite() again, to see if it can be "optimized", in a way that bytes and str would be handled transparently. Well, it's already optimized, and that optimization depends on `bytes`.
6. But well, then there's an option to introduce .awritestr(), and put conversion there. This way, both uasyncio and picoweb need to be patched now, but at least it's a scalable solution reusable for other cases.

pfalcon, 2019-06-13:
An interesting solution would be to have (a null) .encode() method on `bytes`, and call `buf = buf.encode()` unconditionally in .awrite(). That's actually how MicroPython works if built without MICROPY_PY_BUILTINS_STR_UNICODE defined. But otherwise, I saw (ugly, CPython-ridden) code which uses presence of encode/decode methods for ducktyping checks, and that's why with  MICROPY_PY_BUILTINS_STR_UNICODE, there're separate method tables, even if they differ only in encode vs decode.

pfalcon, 2019-06-13:
So, overall, .awritestr() seems like the best solution.

Comments and criticism are welcome though.

tarakanov, 2019-06-14:
Right. I just want to say that I am not so experience in python yet. Now I am working on project which will use this fix. I will try to imlement it and share results.

pfalcon, 2019-06-15:
> Right. I just want to say that I am not so experience in python yet. 

Yes, sometimes it's not easy to decide which is the best way to fix some issue. Anyway, I went for the solution described above and implemented it in picoweb 1.6. I've also added example to test Unicode rendering functionality: https://github.com/pfalcon/picoweb/blob/master/examples/example_unicode.py .

Hopefully, this fixes issues on your side too. If not, feel free to report them.

closed 2019-06-15 by pfalcon