pfalcon/picoweb #53: Picoweb request doesn't return all form data

git-n-pissed, 2019-07-07:
The function read_form_data() calls read() which can result in only part of the expected data being returned.  Seems like it should be calling readexactly() instead.  The problem is described in the micropython forum in detail:

https://forum.micropython.org/viewtopic.php?f=16&t=6647

pfalcon, 2019-07-08:
Good catch. Fixed/new version released.

closed 2019-07-08 by pfalcon

FightMan01, 2021-08-04:
I'm still experiencing this issue.

```js
        function setwifi() {
            obj = document.getElementById("networks")
            var ssid = obj.options[obj.selectedIndex].value;
            password = document.getElementById("password").value;
            $.ajax({ 
                url: '/configure', 
                data: {'password' : password, 'ssid' : ssid}, 
                method: 'POST', 
                success: function(data) { 
                    window.location = data
                } 
            });
        }
```
Only `password` will be in the form. Doing `alert` in the js will give me the correct form.
Python code:
```py
@app.route("/configure")
async def wifi_config(req, resp):
    await picoweb.start_response(resp)
    await req.read_form_data()
    print(req.form)
```