pfalcon/picoweb #56: app.run() fails with TypeError: function expected at most 3 arguments, got 4

xsteadfastx, 2019-08-02:
i try to get picoweb running on the esp32. installed it with `upip.install("picoweb")` and upip.install("micropython-ulogging")`. i used the example code and cant get it running. no matter what i do, i run into this error:

```
>>> upip.install("picoweb")
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Installing picoweb 1.7.1 from https://files.pythonhosted.org/packages/1b/4f/f7d35f90521e95d9d2307f69ff523133d7d4dd6da7ce1ce0c8382e7255fa/picoweb-1.7.1.tar.gz
Installing pycopy-uasyncio 3.1.1 from https://files.pythonhosted.org/packages/5f/24/fb08acdd7ebf1626dcdb3cdaf0d3f463c254a4a3aa2cab70b0ee6562a83d/pycopy-uasyncio-3.1.1.tar.gz
Installing pycopy-pkg_resources 0.2.1 from https://files.pythonhosted.org/packages/05/4a/5481a3225d43195361695645d78f4439527278088c0822fadaaf2e93378c/pycopy-pkg_resources-0.2.1.tar.gz
Installing pycopy-uasyncio.core 2.3 from https://files.pythonhosted.org/packages/6a/96/80a86b1ea4e2b8c7e130068a56f9e8b5bbb28369e48de56a753778e14faf/pycopy-uasyncio.core-2.3.tar.gz
>>> upip.install("micropython-ulogging")
Installing to: /lib/
Installing micropython-ulogging 0.3 from https://files.pythonhosted.org/packages/49/4f/fd39c77fb90dfd188e3e7716556bb3e211edf132d855d7860a8d835b9fbc/micropython-ulogging-0.3.tar.gz
>>> import picoweb
>>> app = picoweb.WebApp("foo")
>>> app
<WebApp object at 3ffd0080>
>>> app.run()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/picoweb/__init__.py", line 299, in run
  File "/lib/uasyncio/core.py", line 163, in run_forever
  File "/lib/uasyncio/core.py", line 129, in run_forever
  File "/lib/uasyncio/__init__.py", line 31, in add_reader
TypeError: function expected at most 3 arguments, got 4
```

i know... no routes defined but i get the same error all over the place... even with routes and stuff. so i thought this might be the most simplest code to get something running. any ideas?

xsteadfastx, 2019-08-03:
i browsed through the last commits of picoweb and read about the change from micropython-uasyncio to pycopy-uasyncio in the setup.py file. i deleted uasyncio from lib on the board and did a `upip.install("uasyncio")`, ran `app.run()` and it worked.

pfalcon, 2019-08-03:
Great. README also should contain the up to date instructions/information, so just reading it should be enough.

closed 2019-08-03 by pfalcon

xsteadfastx, 2019-08-03:
> Great. README also should contain the up to date instructions/information, so just reading it should be enough.

sorry for my question. do you mean the part that it should be used with pycopy and not micropython mainline?

pfalcon, 2019-08-03:
And about pycopy-ulogging or pycopy-logging packages, etc.

xsteadfastx, 2019-08-03:
> And about pycopy-ulogging or pycopy-logging packages, etc.

so if i should have flashed my esp32 with pycopy, a `upip.install("picoweb")` should have worked? i mean after installing logging etc...

im really new to the python on microcontroller world and just read about your fork.

pfalcon, 2019-08-03:
Picoweb is a part of Pycopy project and works only with its infrastructure. I personally don't have esp32, so can't give 100% assurance that everything works seamlessly there (but I don't see why it wouldn't either).

xsteadfastx, 2019-08-03:
> Picoweb is a part of Pycopy project and works only with its infrastructure. I personally don't have esp32, so can't give 100% assurance that everything works seamlessly there (but I don't see why it wouldn't either)

ok... im sold :) will try it. thank you for your amazing work!