pfalcon/picoweb #57: Picoweb.run error

mauroesposito, 2019-09-03:
Using Picoweb on ESP32 devkit v1: one months ago on same board, all ok; tryed on other 2 identhical board, same error.

Here my simple code to test:

`import picoweb

app = picoweb.WebApp(__name__)

@app.route("/")
def index(req, resp):
    yield from picoweb.start_response(resp)
    yield from resp.awrite("Hello world from picoweb running on the ESP32")

app.run(debug=True, host="192.168.2.41")
`

Error after end of booting:

` ...
I (2373) network: CONNECTED
I (5213) event: sta ip: 192.168.2.41, mask: 255.255.255.0, gw: 192.168.2.1
I (5213) network: GOT_IP
:: connecting to network "<<<my_ssid> (try 2)"...
:: WLAN connected to ip 192.168.2.41 gateway 192.168.2.1
I (13773) modsocket: Initializing
* Running on http://192.168.2.41:8081/
Traceback (most recent call last):
  File "main.py", line 12, in <module>
  File "/lib/picoweb/__init__.py", line 302, in run
  File "/lib/uasyncio/core.py", line 129, in run_forever
  File "/lib/uasyncio/__init__.py", line 31, in add_reader
TypeError: function expected at most 3 arguments, got 4
MicroPython v1.11-269-gb29fae0c5 on 2019-09-03; ESP32 module with ESP32
Type "help()" for more information.
`

May be a bug?

Thank you,
Mauro

pfalcon, 2019-09-03:
Not enough information provided, but looks like duplicate of https://github.com/pfalcon/picoweb/issues/45

closed 2019-09-03 by pfalcon

mauroesposito, 2019-09-03:
Ok @pfalcon ,

I can give you information about versions that worked on previous installation.

Versions where all had worked:
Installing **picoweb 1.7.1** from https://files.pythonhosted.org/packages/1b/4f/f7d35f90521e95d9d2307f69ff523133d7d4dd6da7ce1ce0c8382e7255fa/picoweb-1.7.1.tar.gz
Installing **pycopy-uasyncio 3.1.1** from https://files.pythonhosted.org/packages/5f/24/fb08acdd7ebf1626dcdb3cdaf0d3f463c254a4a3aa2cab70b0ee6562a83d/pycopy-uasyncio-3.1.1.tar.gz
Installing pycopy-pkg_resources 0.2.1 from https://files.pythonhosted.org/packages/05/4a/5481a3225d43195361695645d78f4439527278088c0822fadaaf2e93378c/pycopy-pkg_resources-0.2.1.tar.gz
Installing **pycopy-uasyncio.core 2.3** from https://files.pythonhosted.org/packages/6a/96/80a86b1ea4e2b8c7e130068a56f9e8b5bbb28369e48de56a753778e14faf/pycopy-uasyncio.core-2.3.tar.gz

The new versions:

Installing **picoweb 1.7.2** from https://files.pythonhosted.org/packages/a7/cb/1b48103a22f8d67985af5e30ee157c195d2a37b12864e6e8744ad8e7e1b6/picoweb-1.7.2.tar.gz
Installing pycopy-uasyncio 3.1.2 from https://files.pythonhosted.org/packages/73/ef/f5c8a4a9ed2bd20e615d1c638b0547ef1487e4cb0c11533b94dead9191ee/pycopy-uasyncio-3.1.2.tar.gz
Installing **pycopy-pkg_resources 0.2.1** from https://files.pythonhosted.org/packages/05/4a/5481a3225d43195361695645d78f4439527278088c0822fadaaf2e93378c/pycopy-pkg_resources-0.2.1.tar.gz
Installing **pycopy-uasyncio.core 2.3.1** from https://files.pythonhosted.org/packages/35/29/db2c4a67a1e03bf8db4059dcb7cb8697c4b3e0c09535b6882291af13c064/pycopy-uasyncio.core-2.3.1.tar.gz

and for me it's not clear if the problem is in picoweb or uasyncio.

You need other info?

pfalcon, 2019-09-03:
You must use https://github.com/pfalcon/pycopy . Please read the README.

mauroesposito, 2019-09-03:
All clear, thank you.

m