pfalcon/picoweb #58: picoweb memory leak

t35tB0t, 2019-10-02:
picoweb _handle() method will leak memory (rapidly) when yielded socket reads or writes return exceptions.  Adding read.aclose() and write.aclose() to the exception exit code path in _handle() fixes this neatly.

Note:  as-is, uasyncio does not return socket connection failures/errors.  So, POLL_HUP and POLL_ERR events are not returned from the poller ioctl calls.  As-is, bad sockets will hang picoweb co-routines and leak memory with yielded co-routines that never get data.  Adding POLL_HUP and POLL_ERR to uasyncio poller read and write calls allows modwip.c to return these error events.  However, modlwip.d ioctl really should be returning these unsolicited.

picoweb also does not handle socket errors thrown by uasyncio start_server() s.accept().  This will crash picoweb.  Adding exception handling to re-start the server isn't ideal - its is preferred to add socket accept exception handling within the uasyncio start_server() loop so the server can ignore a bad connection attempt and return to waiting for new connections.

The modifications mentioned above were implemented and demonstrated to dramatically tolerate a high rate of bad socket connections while successfully serving requests to picoweb apps.  Without these changes, a picoweb app and micropython can be crashed in seconds.  Even by simply pressing F5 on a windows explorer browser GET request.

pfalcon, 2019-10-02:
Thanks for your report. Please provide the basic information, like: which version of Pycopy (see README), picoweb, uasyncio, which target, how you installed them, etc. Then exact steps to reproduce. It also seems that you report different issues here, I'd suggest to open separate tickets for them. In either case, please provide separate reproduction case for each issue. You say that you made some changes which supposedly fix the issues you experienced. Feel free to post them as a pull request for review.

t35tB0t, 2019-10-03:
Well... that's a lot of detail for something this simple. I'll start with adding an issue to pycopy/extmod/modlwip.c and we can work our way up through the revents socket error return path.  I've been primarily working with the micropython repository and STM architectures.   I pulled pycopy:

commit: 39bdad6c66383b6f6a556d44a39ff9e78c379cb1

modlwip.c is the same but you've made changes uasyncio and uasyncio.core which I have not entirely analyzed.  However, uasyncio.start_server() is essentially identical and is the same missing exception handling in the server loop that I found in micropython.  So, when we fix modlwip.c to return errors unsolicited, the server will throw an exception and terminate if a socket is closed between the IORead(s) and s.accept().   

There is absolutely no question that, with the micropython repository, these fixes stop the socket resets from filling up memory and, voila!, no more crashing from the attack below.  We are trying to close on a project this month so I haven't had time to re-do everything with pycopy - however, seeing some of the variations in the pycopy codebase certainly bears more attention.  Nevertheless, pycopy's modlwip is identical and is not returning any socket exceptions to uasyncio because its not being told to.  And uasyncio won't be able handle unsolicited socket exceptions anyhow because it has no exception handling in the server loop - we don't have to run the code to see that.  ;-P

Here's a script which can  be used to stimulate errors in uasyncio server and client apps (picoweb)

[nasty_resets.py.txt](https://github.com/pfalcon/picoweb/files/3684552/nasty_resets.py.txt)

pfalcon, 2019-10-05:
> that's a lot of detail for something this simple

It's not *that* simple as it seems. First of all, the reference platform for Picoweb is Unix port of Pycopy. Any "doesn't work" claims should be backed by reproduction cases against Unix port. Otherwise, as you noticed, baremetal ports' socket implementations have various issues, actually, there's no baremetal platform which works on par with the reference platform. So again, reproducing against Unix port is important.

> However, uasyncio.start_server() is essentially identical and is the same missing exception handling

This part, I agree with, indeed, it can lead to socket leaks in case of error. I'm ready to fix it (by adding missing exception handling, not in other magic way), just need to validate the fix against integration tests like https://github.com/pfalcon/pycopy-lib/tree/master/uasyncio/benchmark .

> Nevertheless, pycopy's modlwip is identical

Right, within the Pycopy project, I'm trying to "finish" the reference port, Unix one, and leave baremetal ports to the upstream for the time being. I saw that you posted a ticket there (and if you look thru the bug tracker, you'll find more), and that's the route I'd suggest, as I probably won't have any time soon to look into it.

pfalcon, 2019-10-05:
> Here's a script which can be used to stimulate errors in uasyncio server and client apps (picoweb)

Thanks, will give it a try (will run against Picoweb running in Pycopy Unix port, per above).

pfalcon, 2019-10-05:
Some more comments:

> Adding read.aclose() and write.aclose() to the exception exit code path in _handle() fixes this neatly.

It should be quote easy to see (https://github.com/pfalcon/picoweb/blob/master/picoweb/__init__.py#L209) that there's such handling already. `close` defaults to True, so unless a handler asked to keep connections persistent, it's closed, was there exception or not. (And yeah, StreamReader and StreamWriter refer to the same underlying socket, so implementations optimize to closing just one, but convention that's StreamWriter).

> socket accept exception handling within the uasyncio start_server() loop so the server can ignore a bad connection attempt and return to waiting for new connections.

Can you tell which error is just "bad connection" and which "irrecoverably bad system state"? I can't, and as an author of a generic middleware library, I'm not going to add code which effectively masks the latter condition behind user's back, making application go unresponsive without any obvious reason or indication. This boils down to a well-known fact that there's no single error handling strategy which can satisfy all usecases. So, generic middleware libs should do ~ none of it, and instead let user application do the error handling it needs/want. As I mentioned, I agree that such cases should include proper finalization in case of errors, and going to fix that, thanks for the report.

t35tB0t, 2019-10-06:
@pfalcon -  The published code crashes within seconds.  This is a critical usefulness bug and not a matter style.  My patched code runs for days with zero memory issues.  Published code crashes nearly instantly.  But its not my repository - I'm just alerting to the hole I found in an otherwise wonderful piece of work.  Its a tiny hole but it is also a shot right thru the core usefulness  of the system.

The comments above read like I haven't stated my middleware exception handling well enough:

1) uasyncio is actually not floating up *any* modlwip ioctl poll errors.  That's one of the fatal bugs.

2) uasyncio's start_server can throw an exception if it tries to accept a socket which was just reset or otherwise closed.  That's another one of the fatal bugs.

3) picoweb's hande() function has defective exception handling.  This is revealed when it actually has to handle the HUP and ERR events returned from uasyncio.  This is another fatal bug.

The published code with user-implemented socket and connection probing actually only works at all because it usually wins error event race conditions.  That's not good enough for any of us.

At this point, I have a very solid modified system running on my bench.  This morning, I pulled an original set of these modules and I'm adding the minimum mods needed to implement the fixes without the stylistic tweaks and polish I was doing out of habit.  I'll post a zip with a complete running micro-suite which makes this all crystal clear instead of sending a bunch of disjointed diffs.

t35tB0t, 2019-10-06:
Here's some simple testing tools crafted generate abandoned connections, reset connections etc.  The sock_test.py script should be launched as a multiple subprocesses.  This will generate a flood of overlapping reset HTTP connections.  With randomized delays and multiple generators, all the code race conditions are eventually hit.  To simulate the Wifi dropped connections, the scapy library is used with its lower level packet manipulations.  The command-line launch info is commented in the scapy-cAbandon.py script.  One of the options here will provide a simple SYN flood attack.  But I suggest using the -v True mode for our purposes here.

[scapy-cAbandon.zip](https://github.com/pfalcon/picoweb/files/3694170/scapy-cAbandon.zip)
[socket_test.zip](https://github.com/pfalcon/picoweb/files/3694173/socket_test.zip)

t35tB0t, 2019-10-06:
Here's a complete working test case with enhanced exception handling.  Note that extra poll flags are included to force modlwip.c ioctl poll to return the POLL_HUP and POLL_ERR revents just in case you haven't re-compiled a fixed modlwip.c yet.  This makes it "just work" in either case but should be considered debug code and removed at some point.  The test case below was run on a STM32F767.  Prior testing has been performed with a STM32F429.  Scripting and object coding errors were correctly trapped and reported from all layers.  Socket state errors were correctly handled (AFAICT) for floods of connection abandons, resets, and simple SYN attacks.  In all cases browser was able to access the demo webapp GET / info page unless the stack was "jammed" where it would then wait until the nasty connections flood subsided and then would resume.  The memory stats reporting isn't high-fidelity but it does at least show that there isn't any leakage after a lengthy barrage of bad connection behaviors.  Compared to where we started, this works for me and IMHO it addresses the popular WiFi dropout issue (at least from the aspect of not chewing up memory and crashing).  If we have any other connection states we need to consider, let me know.

[PYBFLASH_20191006.zip](https://github.com/pfalcon/picoweb/files/3694185/PYBFLASH_20191006.zip)

P.S.  I apologise if there are any issues or offense with the copyright clauses or other header fluff - this is just text/example code.

pfalcon, 2019-12-20:
I have to admit, there's a bit of issue with project organizations: while some basic principles of the Pycopy project (of which Picoweb is a part) should be obvious to any Pycopy user who follows the project for some time. But it's apparently not so for a newcomers.

This is again the issue I have to admin and I'm sorry for confusion. The guidelines for issue reporting are now spelled out explicitly: https://github.com/pfalcon/picoweb#reporting-issues .

This report, while provides wealth of the information, doesn't adhere to pp.1-3 of the guidelines:

> 1. Please be considerate of the overall best practices and common pitfalls in reporting issues, this document gives a good overview: How to Report Bugs Effectively https://www.chiark.greenend.org.uk/~sgtatham/bugs.html
> 2. The reference platform for picoweb is the Unix port of Pycopy. All issues reported must be validates against this version, to differentiate issues of picoweb/uasyncio from the issues of your underlying platform.
> 3. All reports must include version information of all components involved: Pycopy, picoweb, uasyncio, asyncio.core, any additional modules. Generally, only the latest versions of the above are supported (this is what you get when you install/reinstall components using the upip package manager). The version information are thus first of all important for yourself, the issue reporter, it allows you to double-check if you're using the outdated or unsupported component.

In current form it's not actionable, so I'm closing it. I'm sure we'll return to it when p.5 from the guidelines above is materialized.

closed 2019-12-20 by pfalcon

t35tB0t, 2019-12-24:
Paul-

Yes, we're a newcomer to this tiny project.  Actually, just passing 
through with no present intent to use Pycopy as a whole. No desire to 
author or publish anything.  Upon a simple visual inspection, Pycopy 
code has severe fatal  "race conditions" that will leak memory or crash 
servers (regardless of the port).  You are returning from yielded socket 
calls to uasyncio and then using that same socket object with the naive 
assumption that that socket object is still valid.  Testing that results 
in a socket reset or close between the return from yield and use of that 
socket with throw an ugly exception.  Its not debatable; it has fatal 
server or system crash effect on uasyncio and PicoWeb.  No further 
testing needed to understand that.  Its obvious to any programmer with 
experience and an open mind.

We figured we'd leave a note to help you a little bit.

Cheers and Merry Christmas!

-t