pfalcon/picoweb #60: Incorrect uasyncio function call awitestr()

peterschlem, 2019-12-11:
In __init__.py line 252 an incorrect call to a non existent function **awritestr**
    _yield from writer.awritestr(s)_
Should be 
    _yield from writer.awrite(s)_

pfalcon, 2019-12-11:
Why do you think it's non existent?

peterschlem, 2019-12-11:
I did not indicate that the call was non-existent, I am pointing out that it is a typo: awritestr should be awrite

from Peter Schlemowitz tablet.

peterschlem, 2019-12-11:
To clarify **nonexistent** the stream write function _awritestr_ is not in the _uasyncio_ version 2 library.

peterschlem, 2019-12-12:
There is a **severe memory leak**, about 8K per request. This can be resolved by adding _gc.collect()_ in each routing function.

A _picoweb_ app can be run in a thread (esp32) but the above memory fix still applies.

pfalcon, 2019-12-20:
Please see https://www.chiark.greenend.org.uk/~sgtatham/bugs.html

peterschlem, 2019-12-20:
Read all my comments.
I fixed the problem by implementing the changes as stated - this is not hokum - the program failed when loading a template where 

The awritestr is not implemented in the current uasyncio library but the memory leak is a little harder to diagnose as the initializing uasyncio loop initilization appears to be only called once at startup of the picoweb server.

Peter Schlemowitz
cell: 917 817-7396

pfalcon, 2019-12-29:
> non existent function awritestr

This is incorrect, it's right there: https://github.com/pfalcon/pycopy-lib/blob/master/uasyncio/uasyncio/__init__.py#L202

Please don't report multiple issues in one ticket.

All reports must follow the guidelines: https://github.com/pfalcon/picoweb#reporting-issues

closed 2019-12-29 by pfalcon