pfalcon/picoweb #64: Exception on ESP8266

wesparish, 2020-02-27:
I'm not sure if this is a picoweb, uasyncio, or pycopy problem, leaning more towards pycopy (specifically micropython code), but unfortunately, I'm new to the project and not familiar at all with the sockets stack.

When starting a picoweb server, I'm getting the following exception in the uasyncio thread:
```
DEBUG:main:network config: ('172.16.1.127', '255.255.255.0', '172.16.1.1', '172.16.1.222')
* Running on http://0.0.0.0:8081/
Traceback (most recent call last):
  File "main.py", line 50, in <module>
  File "picoweb/__init__.py", line 311, in run
  File "picoweb/__init__.py", line 293, in serve
  File "uasyncio/core.py", line 116, in run_forever
  File "uasyncio/__init__.py", line 266, in start_server
OSError: [Errno 11] EAGAIN
```

This seems to work about 50% of the time if I put an arbitrary sleep (tested with 3s) after my network connection code, so perhaps it's some kind of race condition?

pycopy firmware was built with additional modules using:
```
wes@w-dock4:~/pycopy/ports/esp8266$ ../unix/pycopy -m upip install -p modules picoweb
Installing to: modules/
Warning: pypi.org SSL certificate is not validated
Installing picoweb from https://files.pythonhosted.org/packages/ba/e4/68983f6150f7fbd98b6b937bf7f76aeb4eb9ba4abb7c93dc05b2ccdbd25c/picoweb-1.8.1.tar.gz
Installing pycopy-uasyncio from https://files.pythonhosted.org/packages/3c/1e/e0953531408f1d237456bcb324b14c1b84218ad7ee5702a3022e7cde7a71/pycopy-uasyncio-3.5.tar.gz
Installing pycopy-pkg_resources from https://files.pythonhosted.org/packages/05/4a/5481a3225d43195361695645d78f4439527278088c0822fadaaf2e93378c/pycopy-pkg_resources-0.2.1.tar.gz
Installing pycopy-uasyncio.core from https://files.pythonhosted.org/packages/ca/b2/c5bba0bde7022b6d927a6144c026d7bf310d3f8a20e031571fbf1a08a433/pycopy-uasyncio.core-2.3.2.tar.gz
```

Any help would be greatly appreciated! Thanks!

pfalcon, 2020-02-29:
Please see https://github.com/pfalcon/picoweb#reporting-issues, especially relevant pp.2, 3. I can however say that there were reports similar to this some time ago, and the upstream, which maintains esp8266 port, made some fixes in that regard. So, please make sure you're using the latest Pycopy version.

wesparish, 2020-02-29:
Thanks, I wasn't sure if you had an esp8266 to test against to help track down the issue. I can keep digging, and I also ordered an esp32 to see if it's maybe an issue with low resources on the current board.

I did see the issue you were mentioning above, but was unable to git revert that specific change as there was another commit afterwards to "fix" the issue. No clue if that specific accept issue is truly fixed.

Yes, I cloned Pycopy fresh this week, so it should be up to date. Thanks again for your work on Micropython, Pycopy and these libraries, they are truly appreciated for us hobby users!

pfalcon, 2020-03-03:
> Thanks, I wasn't sure if you had an esp8266 to test against to help track down the issue.

Sorry, I have a long list of things to investigate and fix, and for the time being work on generic language-level things rather than hardware-specific issues. I'd suggest trying to prepare a minimal testcase showing the issue (with bare usocket/uselect code) and submitting upstream with instructions on how to reproduce.

closed 2020-03-03 by pfalcon

cmpadden, 2020-03-27:
@wesparish, I'm aware that this issue is closed at this point, but I wanted to let you know that I had experienced a similar issue while using MicroPython v1.12, after installing `picoweb` via `upip.install('picoweb')`. And it was resolved by installing the latest version of `uasyncio` via `upip.install('uasyncio')`.

I'm not sure how this differs from the `pycopy-uasyncio` dependency listed in the _setup.py_ file, but it worked for me -- hope this helps in some way.