pfalcon/picoweb #65: Running Picoweb on micropython

JLFra, 2020-03-28:
Hello,
I found a solution to run Picoweb on micropython esp32 wroom 32D
Here:
https://forum.micropython.org/viewtopic.php?f=18&t=8048&start=10
But I don't understand why it's runnig only on firmware v1.11?
Thanks to help me.
Regards.
Julien

pfalcon, 2020-04-06:
Please see Picoweb README, https://github.com/pfalcon/picoweb , for requirements to run Picoweb.

closed 2020-04-06 by pfalcon

JLFra, 2020-04-06:
Hello,

I can't see in the README what I have found for MicroPython. Why don't you want to add in the readme. And why it don"t run only under V.1.1 firmware of micropython? Thanks to be very precise in your answer. And great for your Framework Picoweb.

pfalcon, 2020-04-06:
Picoweb requires [Pycopy](https://github.com/pfalcon/pycopy), and this is clearly stated in the README. Working with any other Python implementations is not guaranteed, and not supported by me due to lack of resources (but see p.5 in https://github.com/pfalcon/picoweb#reporting-issues).