pfalcon/picoweb #67: Larger number of parallel connections fail

ctschach, 2020-07-09:
Hi,

I've created a little web app. When you load the page for the first time there are 8 file loaded:

index.html
2x css files
2x js files
3x images

The browser only succeeded on 4 connections. For the other remaining requests the connection is accepted but then instantly closed before the browser could even send his request. I checked this via tcpdump.

Is there a way to limit the number of parallel connections that picoweb accepts? Instead of accepting and instantly closing the connection it would be helpful to just wait until "connection slots" are free. BTW, you don't even see the failed connections in the debug log.

pfalcon, 2020-07-10:
See https://github.com/pfalcon/picoweb/issues/66#issuecomment-656615876 . Additionally: https://www.chiark.greenend.org.uk/~sgtatham/bugs.html

JK-de, 2022-07-17:
the uasyncio-function 'start_server' handles by default only 5 connections. You can change line 303 in `picoweb/__init__.py` to
```python
loop.create_task(asyncio.start_server(self._handle, host, port, backlog=10))
```
ich you have enough memory to handle 8 connections at the same time