pfalcon/picoweb #70: How do I install picoweb?

aguaviva, 2020-11-12:
I haven't found any instructions 

Cheers

aguaviva, 2020-11-12:
1) Install dependencies:
```
import upip
upip.install('micropython-uasyncio')
upip.install('micropython-pkg_resources')
upip.install("micropython-ulogging")
```

2) copy the picoweb folder into the microcontroller

3) In the examples, remember to add your host ip to the app.run line:
`app.run(debug=True**, host = "192.168.1.5"**)`

pfalcon, 2020-11-12:
> I haven't found any instructions

Try README (including links in it). But in either case, you install Picoweb as any other Pycopy package. There's documentation section on that: https://pycopy.readthedocs.io/en/latest/reference/packages.html

closed 2020-11-12 by pfalcon